microsoft/Azure_Kinect_ROS_Driver #199: incorrect intrinsic calibration without an option to update

christian-rauch, 2021-07-08:
**Describe the bug**
The intrinsic calibration that is used to project colour and depth pixels to a 3D point cloud is wrong or not properly processed. According to https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/726 and https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/105 it is neither possible to update the intrinsic parameters on the camera, nor does the ROS node support calibration.

**To Reproduce**
Steps to reproduce the behavior:
1. set up a scene with AprilTags
2. visualise the point cloud as well as the estimated tag frames in RViz
3. see that frames do not align with depth

**Expected behavior**
The intrinsic calibration should provide proper 3D point clouds.

**Screenshots**

The following images show a point cloud together with the coordinate frames of two AprilTags in RViz. One AprilTag is flat on the table and another one is attached to the side of box and thus orthogonal to the table.

With the Kinect factory calibration, those points and coordinate frame do not align:
[image]
[image]

As a reference, here is the data from an Asus Xtion Pro Live using the `openni2_camera` package and [manual intrinsic calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration). As you can see, these frames perfectly align with the point cloud.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

ooeygui, 2021-07-08:
Thank you for the detailed writeup. 
This thread (https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/83) has a discussion about calibration. Color and depth are handled separately, each with their own calibration. Since most downstream tooling doesn't seem to support the required calibration, we've discussed predistorting to align. However, this would introduce performance issues. We have not reevaluated calibration with the switch to ROS2.

christian-rauch, 2021-07-08:
I am specifically talking about the intrinsic calibration of the camera (a.k.a. the camera matrix K or C) here, not the extrinsic calibration between two Kinects or between the colour and depth sensor. The colour <-> depth alignment seems to be ok as I don't see any issues with mismatching colour/depth values on edges etc. But the intrinsic calibration for projecting a point from 2D to 3D and vice versa seems to be off.

From the images above, it is clear that at least one of the cameras (colour or depth) is using the wrong intrinsic parameters. But I cannot know if I can trust the AprilTag pose (i.e. the colour intrinsics are correct) or the depth (i.e. the depth intrinsics are correct) or neither of those (as I don't have a third reference).

jmachowinski, 2021-08-05:
Hi @christian-rauch how are you ? Hope fine ;-)

Did you find out more details, what is off ? 
Or asked the other way around, did your test show anything weird with the depth intrinsic ?

christian-rauch, 2021-08-05:
@jmachowinski Nice to see you :-) It's a small (robotics) world.

tl;dr: The factory calibration is actually correctly used by ROS. I just thought that the driver already publishes rectified images.

When I created this issue, I thought that the driver is already publishing rectified images using the factory calibration. But I was wrong about this: The `image_raw` topics indeed are the raw unrectified images and you have to use the `image_proc`nodelets to rectify the RGB and IR images or do the projections yourself.

I created https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/200 to enable a manual calibration similar to the OpenNI drivers. In https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/200#issuecomment-883406304 I compared the raw and rectified images (factory and manual). As you can see from the lights in the ceiling, the factory calibration already provides very good rectification, although with a black border. The manual calibration keeps the image boundaries, but then does not achieve the same level of rectification. But maybe this can also be fixed with a much larger calibration board.

I am using the manual calibration because we are processing the colour images directly and don't want to deal with cropping out the black border. If you use the colour image as part of a 3D-only pipeline, then using the factory calibration is better.

jmachowinski, 2021-08-06:
Thanks for the update.

So the title of the issue is currently a bit misleading. The intrinsics are correct, but there is
no option, to give an own calibration (if you want that for reasons...).

christian-rauch, 2021-08-06:
> So the title of the issue is currently a bit misleading. The intrinsics are correct, but there is
> no option, to give an own calibration (if you want that for reasons...).

Right, I did not update the description after opening PR #200. I updated the title just now and added a clarification to the issue description.

closed 2021-11-19 by christian-rauch