microsoft/Azure_Kinect_ROS_Driver #201: Failed to transform depth image to point cloud!

wangmiaowei, 2021-07-10:
When I run the ROS, I get something as follows:
[ WARN] [1625914072.118329000]: Initializing the device to realtime offset based on wall clock: 1625914071552925200 ns
[2021-07-10 18:47:52.195] [error] [t=4291] ../src/transformation/transformation.c (584): k4a_transformation_t_get_context(). Invalid k4a_transformation_t (nil)
[2021-07-10 18:47:52.195] [error] [t=4291] ../src/transformation/transformation.c (873): Invalid argument to transformation_depth_image_to_point_cloud(). transformation_handle ((nil)) is not a valid handle of type k4a_transformation_t
[2021-07-10 18:47:52.195] [error] [t=4291] ../src/sdk/k4a.c (1286): transformation_depth_image_to_point_cloud(transformation_handle, depth_image_buffer, &depth_image_descriptor, camera, xyz_image_buffer, &xyz_image_descriptor) returned failure in k4a_transformation_depth_image_to_point_cloud()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to transform depth image to point cloud!
[azure_kinect_ros_driver-3] process has died [pid 4023, exit code -6, cmd /mnt/f/Ubuntu_win/Project/AzureROS_ws/devel/lib/azure_kinect_ros_driver/node __name:=azure_kinect_ros_driver __log:=/home/miaoweiw/.ros/log/ef8721d2-e16b-11eb-bfcd-00155d704bc9/azure_kinect_ros_driver-3.log].
log file: /home/miaoweiw/.ros/log/ef8721d2-e16b-11eb-bfcd-00155d704bc9/azure_kinect_ros_driver-3*.log
^C[robot_state_publisher-2] killing on exit
[joint_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ooeygui, 2021-07-12:
Hi there,
This error is coming from the Azure Kinect SDK. @qm13 do you have a recommendation?

ooeygui, 2021-07-13:
Are you using a default launch file or have you customized it?

sssphil, 2022-04-17:
Hi @ooeygui, 
I've run into the same problem when converting depth to color. I copied 'driver.launch' and gave it a test mkv file.
I'm running on WSL. Maybe this is related to [https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/810](url) [https://github.com/etiennedub/pyk4a/issues/78](url)