microsoft/Azure_Kinect_ROS_Driver #210: Driver hangs forever if initial open failed

jmachowinski, 2021-08-04:
**Describe the bug**
If at the end of constructor of K4AROSDevice k4a_device_ is still nullptr,
a ROS_ERROR is outputted, but the node is not shut down. This leads to
a 'hanging' node.
This error occurres, if multiple drivers are launched at the same time.

As every driver claims the device exclusive then checks the serial, and afterwards
releases the device if the serial is not the correct one, it blocks other processes 
from opening the device during this time. The second driver will then skip the device,
as k4a::device::open failed, and end up with no device at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Have multiple cameras connected
2. Start multiple nodes in parallel
3. Repeat until one driver gives the output 'Failed to open a K4A device. Cannot continue.'

**Expected behavior**
In this case the node should shut down.

Note, I got a merge request ready for fixing this issue.

christian-rauch, 2021-08-12:
Related issue about opening multiple devices in parallel: https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/97.

jmachowinski, 2021-11-22:
ping ?