microsoft/Azure_Kinect_ROS_Driver #214: depth_registered/points is empty

xingchenzhang, 2021-08-29:
Hi,

I am trying to use depth_registered/points in my application. However, in my case depth_registered/points is empty so it is not working.

Can anyone shed some ideas on this?

Thank you very much!

christian-rauch, 2021-09-16:
Which launch file are you using?

Launching via `driver.launch` provides `/points2` with `sensor_msgs/PointCloud2`. Alternatively, you can use the colour, registered depth image and the intrinsic to do the projection yourself.

HilbertXu, 2022-03-29:
> Which launch file are you using?
> 
> Launching via `driver.launch` provides `/points2` with `sensor_msgs/PointCloud2`. Alternatively, you can use the colour, registered depth image and the intrinsic to do the projection yourself.

Hi Christian,

We are using the kinect_rgbd.launch to launch the Azure kinect and the depth_registered/points topic is empty, could you please give us some ideas to solve this?

Many thanks!

christian-rauch, 2022-03-30:
I download `OFFICE_SAMPLE_NFOV_UNBINNED.zip` from https://www.microsoft.com/en-us/download/details.aspx?id=58385 and ran the driver launch file via:
```sh
export MKV_PATH=$HOME/Downloads/OFFICE_SAMPLE_NFOV_UNBINNED/OFFICE_SAMPLE_NFOV_UNBINNED.mkv
roslaunch azure_kinect_ros_driver driver.launch recording_file:=$MKV_PATH recording_loop_enabled:=true depth_mode:=WFOV_UNBINNED
```
The driver then publishes a colour point cloud as `sensor_msgs/PointCloud2` on topic `/points2`. If you only want the registered colour point cloud, this should be sufficient.

When I use `kinect_rgbd.launch` instead, then the topic `depth_registered/points` does not exist:
```sh
$ rostopic list | grep point
/depth/points
/depth_to_rgb/points
```

The `driver.launch` starts a "stand-alone" node just called `node`:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/launch/driver.launch#L50
which is defined in the CMake:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/CMakeLists.txt#L51-L56

The `kinect_rgbd.launch` launches a nodelet manager
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/launch/kinect_rgbd.launch#L75-L80
with the driver nodelet
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/launch/kinect_rgbd.launch#L83-L85
defined in the CMake
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/CMakeLists.txt#L60-L65
and the `rgbd_launch` nodelets:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/c0742b9e470c9e688d796029f10cb52e1a763a4a/launch/kinect_rgbd.launch#L109

christian-rauch, 2022-03-30:
@xingchenzhang @HilbertXu If you still want to use the `kinect_rgbd.launch` launch file (e.g. for a manual calibration), you have to apply PR #242 .

After this, point clouds will be published on `/depth/points` (points only) and `/depth_to_rgb/points` (coloured points). Note that the registered point cloud in `/depth_to_rgb/points` is published at a lower frequency due to the overhead. So if you are happy with the factory calibration, you should use `/points2` via `driver.launch` intead.

HilbertXu, 2022-03-31:
> @xingchenzhang @HilbertXu If you still want to use the `kinect_rgbd.launch` launch file (e.g. for a manual calibration), you have to apply PR #242 .
>
> After this, point clouds will be published on [...]

Hi Christian,

Many thanks for you help, it's super useful and detailed!
I will test it soon : )

PR #242 (2022-04-19): set 'depth_registered_filtered' to the same value as 'depth_registered'

## Fixes #214 

### Description of the changes:
- set `depth_registered` and `depth_registered_filtered` to the same value (`depth_to_rgb`)

### Required before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/master/docs/building.md) locally
- [x] I tested my changes with a device*
- [ ] I changed both the ROS1 and ROS2 branches

* tested with  [... cut]
closed 2022-04-19 by ooeygui