microsoft/Azure_Kinect_ROS_Driver #215: Failed to poll cameras: node cannot continue.

ymxlzgy, 2021-08-31:
**Describe the bug**
When I launch the kinect_rgbd.launch, this error has shown up. But the driver.launch is fine.

**Logs**
[ WARN] [1630399140.685111182]: Camera calibration file /home/edge/.ros/camera_info/azure_kinect_ir_000371312112_NFOV_UNBINNED.yaml not found.
[ WARN] [1630399140.991443637]: Initializing the device to realtime offset based on wall clock: 1630399140791121325 ns
[ WARN] [1630399141.165516044]: Initializing or re-initializing the device to realtime offset: 1630399140814935623 ns
[ WARN] [1630399141.192361610]: Image processing thread is running behind.
Expected max loop time: 0.033333333
Actual loop time: 0.500916938

[ WARN] [1630399146.805491469]: Initializing or re-initializing the device to realtime offset: 1630399140825394300 ns
[ WARN] [1630399146.829643460]: Initializing or re-initializing the device to realtime offset: 1630399140813543617 ns
[ WARN] [1630399151.239423829]: Image processing thread is running behind.
Expected max loop time: 0.033333333
Actual loop time: 0.044799018

[ WARN] [1630399152.387944799]: Initializing or re-initializing the device to realtime offset: 1630399140827120107 ns
[ WARN] [1630399152.497890270]: Initializing or re-initializing the device to realtime offset: 1630399140816968248 ns
[FATAL] [1630399154.612585219]: Failed to poll cameras: node cannot continue.
[k4a_nodelet_manager-4] process has finished cleanly
log file: /home/edge/.ros/log/e01bc812-0a36-11ec-9b2e-509a4c5540ba/k4a_nodelet_manager-4*.log

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04

zhang159560293, 2024-04-14:
> Did you solve this problem?

I have solved this issue. Chang a usb3.2 gen2 line. It is because of usb port or usb line.

nyxrobotics, 2024-04-17:
If nodelet crashes, I can understand that it is a compatibility issue, but if nodelet manager crashes, it is a bug in nodelet, right?

ooeygui, 2024-04-17:
The Azure Kinect DK is very sensitive to the USB host controller and CPU performance. the multiple init and resync suggests a hardware incompatbility.