microsoft/Azure_Kinect_ROS_Driver #227: Fix publishing ordered pointcloud messages as unordered

rockmasstech, 2021-11-10:
### **Problem**
Related to [Issue 163](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/163)

On the latest **melodic branch**, when subscribing to the `/points2` pointcloud topic, the received [`sensor_msgs::PointCloud2`](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/PointCloud2.html) messages are unordered (i.e. height = 1).

In my testing for example, when `point_cloud_in_depth_frame=false` and `color_resolution=720P`, the message will have a height of 1 and width of 921,600 (exactly 720 * 1280). It seems to be publishing an ordered point cloud as if it were unordered.

This is inconvenient - for instance when using `pcl::fromROSMsg()` to convert the message to a `pcl::PointCloud`, the point cloud cannot be grid indexed because it's unordered. This can be worked around by writing a function to manually grid index a point with pointer arithmetic, but certainly not ideal.

### **Requested Solution**
Investigating why the `sensor_msgs::PointCloud2` message are published unordered in the first place, took me to [`K4AROSDevice::fillColorPointCloud()`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L666).

The line [`pcd_modifier.setPointCloud2FieldsByString(2, "xyz", "rgb");`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L683) sets the message's pointcloud fields and also resizes the actual data array. No issue here. Refer to [line 198 in the ROS implementation](http://docs.ros.org/en/melodic/api/sensor_msgs/html/impl_2point__cloud2__iterator_8h_source.html).

However a few lines down [`pcd_modifier.resize(point_count);`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L693) is called, which I believe is redundant, and also assigns height to 1 and assigns width accordingly, making it unordered. Refer to [line 116 in the ROS implementation](http://docs.ros.org/en/melodic/api/sensor_msgs/html/impl_2point__cloud2__iterator_8h_source.html).

Removing this line leads to publishing an ordered point cloud message with the expected height and width.
Tested and working [on my fork](https://github.com/rockmasstech/Azure_Kinect_ROS_Driver/commit/0e672845c7cb9e66bdb9e539299339213b4b04b4).

### **Additional context**
I believe same issue applies to [`K4AROSDevice::fillPointCloud()`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L725).

christian-rauch, 2021-11-19:
Can you send a PR with your changes so they can be reviewed and properly integrated?

ndaley7, 2023-08-31:
> ### **Problem**
> Related to [Issue 163](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/163)
>
> On the latest **melodic branch**, when subscribing to the `/points2` pointcloud topic, the [...]

Thank you for this fix!