microsoft/Azure_Kinect_ROS_Driver #231: Several blue circles appearing on RGB Camera view

gantproductions, 2021-12-20:
Hello all. Please let me know if this belongs in the SDK Repository instead. I was also instructed to tag @qm13  on this :)

I just unboxed 5 new sensors and have them all up and running. One of these sensors however has several blue rings appearing in the RGB camera feed only. They don't seem to be affected by changing the exposure, contrast, or any other settings in Depthkit. I see them appear both in the RGB feed of the official Kinect Viewer Tool and also the RGB feed of depthkit. Let me know if anyone has seen this before or has any ideas on how to fix. I did try and reinstall the firmware and this did not fix it.

Picture below:
[image]

ooeygui, 2021-12-21:
Thank you for the report. This does appear to be an issue with the camera itself; not with the ROS driver or Azure Kinect driver.

ooeygui, 2021-12-30:
I'm going to close this out - I believe this camera needs to be repaired under warrantee.

closed 2021-12-30 by ooeygui