microsoft/Azure_Kinect_ROS_Driver #232: Depthcloud and Pointcloud2 not consistent in Rviz

petpetpeter, 2021-12-23:
I'm trying to convert depth image from depth_to_rgb/image_raw topic into pointcloud2
but when I compared it with the PointCloud data in /points2 topic it was a little bit off.
I try checking this problem in RVIZ

[image]

The RGB pointcloud is from /points2 topic, the red pointcloud is from /depth_to_rgb/image_raw
Is this normal or some problem? Which one is the more accurate position from the rgb_camera_link?

christian-rauch, 2022-04-19:
`/points2` is publishing the internally rectified, registered and back-projected data. `/depth_to_rgb/image_raw` publishes the unrectified image. You have to rectify the image, e.g. via [kinect_rgbd.launch](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/kinect_rgbd.launch).

petpetpeter, 2022-04-20:
Thanks, I launched the camera with driver.launch, that could be the reason.

closed 2022-04-20 by petpetpeter

jiejie567, 2023-02-16:
@petpetpeter  Hi, I think /points2 and /depth_to_rgb/image_raw correspond different coordinate system, which is the result there are some drifts.