microsoft/Azure_Kinect_ROS_Driver #233: Transform function pixel_2d_to_point_3d

ridasaleem0, 2022-01-17:
Hi,
Can we use all the transform sub-functions provided within Kinect SDK using ROS nodes? or just the color to depth image and vice versa? 
I have an algorithm with makes use of transform.pixel_2d_to_point_3d() to get 3d points in mm and I am unable to find anything related to it in the repository.

ooeygui, 2022-01-17:
The Azure Kinect SDK has distro affinity, which has prevented us from migrating the melodic repo forward. The ROS2 Foxy repo can be moved forward to latest SDK. We should branch the Melodic repo into a noetic repo in order to support the latest SDK.

ridasaleem0, 2022-01-18:
@ooeygui Thank you for your quick response. So we cannot currently make use of the latest SDK functions using ROS nodes? Can you maybe provide an alternative?

ooeygui, 2022-01-18:
@ridasaleem0 We have not made the changes in the repo to enable the Kinect ROS node with later ROS distributions. Later distributions of the Kinect SDK require features which are incompatible with Melodic, so require Noetic. In order to support later SDKs, we need to create a noetic branch, update the makefiles, and update the code to address the API changes.

I created this work item - https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/234