microsoft/Azure_Kinect_ROS_Driver #234: Support Noetic with latest Azure Kinect SDK on 20.04.

ooeygui, 2022-01-17:
This is a work item to fork and support Noetic with the latest Kinect SDK.

yuukireina05, 2022-01-25:
Is Azure Kinect SDK going to support Ubuntu 20.04 as well as ROS Noetic? When will the update release?

mac137, 2022-11-27:
The same question as above. Thanks!

dirkmcpherson, 2023-01-10:
Is there a branch somewhere for this? Or does someone have an implementation plan?

Ubuntu 20.04 with Cuda 11.8 on Ros Noetic gives me:

```
Non-zero status code returned while running Pad node. Name:'workaround_cudnn' Status Message: 
2023-01-10 15:41:53.928483133 [E:onnxruntime:K4ABT_ORT_Environment, cuda_call.cc:93 CudaCall] CUDA failure 48: no kernel image is available for execution on the device ; GPU=0 ; hostname=redacted ; expr=cudaGetLastError(); 
```

ooeygui, 2023-01-10:
@dirkmcpherson The ROS driver is dependent on the SDK, which is not officially supported on 20.04 or beyond. Please +1 this issue: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1263

dirkmcpherson, 2023-01-10:
@ooeygui Thanks! 

FYI it seems to work with lib4ka1.4* and libk4abt1.1* if you change the relevant version numbers in the top level CMakeLists.txt and, in addition, fix the error described here: [microsoft/Azure_Kinect_ROS_Driver/issues/143](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/143). 

I can track skeletons on Noetic with this setup.

omerKurkutlu, 2023-07-04:
> Is Azure Kinect SDK going to support Ubuntu 20.04 as well as ROS Noetic? When will the update release?

**I succeeded to run Azure Kinect in ROS Noetic with Ubuntu 20.04.**

First, I followed this [video.](https://www.youtube.com/watch?v=XJQtAod5v2A&t=10s&ab_channel=JFK_Tech)

Then, I got output but when I selected the 3d option on SDK, it was giving an error as in the image.

[image]

To solve this problem, I need to update the OpenGL version from 4.3 to 4.6. (It has to be minimum 4.4)
You can see your version by typing this command in terminal "inxi -G".

[image]

In order to use ROS- Noetic there is no repository but the one in the [link](https://github.com/microsoft/Azure_Kinect_ROS_Driver) for ROS-Melodic works without a problem. You just download the GitHub repo in your ROS-Noetic catkin-ws and catkin_make. That is it...  

[image]