microsoft/Azure_Kinect_ROS_Driver #235: On "color spill", or "flying pixels"

francescomilano172, 2022-02-02:
**Describe the bug**
Using the factory calibration, point clouds exhibit “color spill”, or “flying pixels” at the object boundaries. This happens both with the latest version of the ROS driver and with the Azure Kinect Viewer. Can this problem be solved through manual calibration?

**To Reproduce**

#### Kinect Azure Viewer
1. Launch `k4aviewer` and open the device.
2. As `View Mode` select `3D` and `Color`.
3. Point the camera on any object and observe the color spill at the object boundaries.

#### `/points2` topic from ROS driver
1. Set the flags `point_cloud` and `rgb_point_cloud` to `true` in [`kinect_rgbd.launch`](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/kinect_rgbd.launch). Also keep `point_cloud_in_depth_frame` to `false`, so as to have `depth_to_rgb`-like backprojection. This should be the choice that gives less color spill, as mentioned for instance [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/92).
2. Run `kinect_rgbd.launch` and look at the `/points2` topic in RVIZ.

**Expected behavior**
In the desired setup, I would need to have aligned and rectified RGB and depth frames and use camera intrinsics to retrieve a point cloud. In the point cloud, there should be no color spill.

**Screenshots**
In all the following screenshots, colorspill on the edges of the pink ball can be noticed
- Rviz screenshot of the `/points2` topic (`point_cloud_in_depth_frame` set to `true`):
[image]
- Rviz screenshot of the `/points2` topic (`point_cloud_in_depth_frame` set to `false`):
[image]
- Azure Kinect Viewer screenshot of the 3-D point cloud, 720p RGB, WFOV Depth:
[image]
- Azure Kinect Viewer screenshot of the 3-D point cloud, 720p RGB, NFOV Depth:
[image]

**Desktop:**
 - OS: Ubuntu
 - Version: 18.04.6
 - Commit of Azure_Kinect_ROS_Driver: c0742b9e470c9e688d796029f10cb52e1a763a4a
 - k4aviewer Firmware: RGB Camera: 1.6.102, Depth Camera: 1.6.75

**Additional context**
It is unclear to me whether color spill is a problem that can be avoided. Previous conversations seem to hint at the fact that this is an inherent limitation of the Azure Kinect (e.g., the paper mentioned [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1687#issuecomment-936868543)).
On the other hand, some threads seem to suggest that the problem might be related to imperfect calibration/camera alignment (e.g., [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/83) and [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1058)) and that a custom calibration can yield better alignment between the RGB and IR cameras (e.g., [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/803#issuecomment-555259641) and [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/803#issuecomment-645070253)). However, the conversations are overall inconclusive, with mixed opinions ([negative 1](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/803#issuecomment-570006001), [negative 2](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/865), [positive](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1058#issuecomment-596565013), [unclear](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1201)).

I also saw that there is now the possibility to manually calibrate the intrinsics of the cameras and use them through the ROS interface instead of the factory calibration ([here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/105) and [here](https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/200)). Can a custom calibration alleviate or fix this problem, or is it a hardware limitation?

Also, is this to some extent due to the interpolation that is introduced both when warping the depth image into the RGB frame (e.g., [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2feb3425259bf803749065bb6d628c6c180f8e77/src/sdk/k4a.c#L1188) and [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/2feb3425259bf803749065bb6d628c6c180f8e77/src/transformation/rgbz.c#L258)) and when rectifying the images before backprojection (as happens in `rgbd_launch`/`image_geometry`/`cv2.remap`, see e.g., [this issue](https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/212))?
In particular, different interpolation schemes produce very different results (see, e.g., [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/588#issuecomment-519627762)), but even the recommended nearest-neighbor interpolation which is used in `rgbd_launch` for rectification ([here](https://github.com/ros-drivers/rgbd_launch/blob/2123238c24a4b9a59106f8264893bca9dc278462/launch/includes/depth_registered.launch.xml#L67-L69)) does not solve the color spilling problem.

ooeygui, 2022-02-02:
Thank you for the bug and details.

christian-rauch, 2022-04-19:
If `k4aviewer` already shows this effect on the pointe cloud, then there is nothing that the ROS node can do for the point cloud on `/points2`.

You can use the original colour and depth image and the intrinsics+extrinsics in two ways:
1. use the [rgbd_launch](http://wiki.ros.org/rgbd_launch) nodes via `kinect_rgbd.launch` to do the rectification, registration and projection manually
2. manually calibrate the camera and try if `kinect_rgbd.launch` provides better results afterwards

It may be sufficient to only adjust the extrinsic calibration between the colour and depth camera and keep the intrinsics as they are.

francescomilano172, 2022-04-20:
Hi @christian-rauch, thank you for your answer. For number 1., yes, this is what we are already doing, following the procedure in https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/212. For 2., can you recommend any calibration procedure for the extrinsics between the colour and the IR camera?

christian-rauch, 2022-04-20:
Since https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/200, you can use the [`camera_calibration`](http://wiki.ros.org/camera_calibration) package to assign new intrinsic parameters. You can calibrate the RGB and IR cameras separately.

The extrinsic parameters are published as tf. But I don't know a procedure to does that automatically. You may have to figure that out manually by adjusting the extrinsic tf and comparing the quality of the registration.