microsoft/Azure_Kinect_ROS_Driver #236: How to use ROS-Driver to drive multiple kinects?

2361098148, 2022-02-15:
I tried to drive two kinects simultaneously, how should I do?

ooeygui, 2022-02-15:
The ROS node does not currently support multiple Kinects on one system very well; it can be made to function by having multiple nodes launch each connecting by device ID in the launch file for each ROS node. 
Additionally, the Kinect is very sensitive to the USB host controller and drivers, so each Kinect should be on a separate USB Bus. Between these two facts, when multiple Kinects are required, then multiple computers are recommended.

2361098148, 2022-02-16:
Thanks a  lot for your answer at first, and I  have another two questions according to your reply. Firstly, I stick to open two kinects in one computer, if I need to roslaunch two ROS-Driver simultaneously with the different device ID? I'd appreciate it if you can give me more information such as an example code or instruction. Secondly, it seems like we could use SDK to merge two Kinects' information.  What should I do if I want to accomplish a similar function by ROS-Driver?  If I need to get the relative posture of the Kinects manually and merge the point cloud by ICP algorithm? Is there some example code?

ooeygui, 2022-02-16:
Here is the original feature request to support multiple camera setups, including pointcloud fusion - https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/126.
The [sample launch file](https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/driver.launch) is currently optimized for single instance with no namespace support. It does have parameters `sensor_sn`, which allows you to specify the serial number of a kinect for that specific driver to bind to, and `tf_refix` so you can link each into a urdf. You'll have to use the Kinect SDK to find the serial number for a specific device. In order to launch multiple devices you can import this launch file into another launch file and add a namespace around it.

2361098148, 2022-02-17:
**Thanks a lot! After your instruction, i have tried this way:**
<node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> 

    <param name="tf_prefix"         type="string" value="1" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000466592912" />       
    
    <param name="tf_prefix"         type="string" value="2" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>
I learned it from #97, but I met similar subsequent problems as well. It seemed like both Kinects worked, but only the first Kinect's data could be obtained. It is strange because we could receive the second Kinect's coordinate system in Rviz without more data. So I opened rqt_graph and found that it seemed like that both Kinects published the data on the same topic. So I doubt that if I need to apply distinct topic's name  to different cameras?  Or should I alter the launch file in another way?

2361098148, 2022-02-17:
Thanks a lot! After your instruction, i have tried this way:
<node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> 

    <param name="tf_prefix"         type="string" value="1" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000466592912" />       
    
    <param name="tf_prefix"         type="string" value="2" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>
I learned it from #97, but I met similar subsequent problems as well. It seemed like both Kinects worked, but only the first Kinect's data could be obtained. It is strange because we could receive the second Kinect's coordinate system in Rviz without more data. So I opened rqt_graph and found that it seemed like that both Kinects published the data on the same topic. So I doubt that if I need to apply distinct topic's name  to different cameras?  Or should I alter the launch file in another way?

2361098148, 2022-02-17:
**Thanks a lot! After your instruction, i have tried this way:**
<node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> 

    <param name="tf_prefix"         type="string" value="1" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000466592912" />       
    
    <param name="tf_prefix"         type="string" value="2" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>
I learned it from #97, but I met similar subsequent problems as well. It seemed like both Kinects worked, but only the first Kinect's data could be obtained. It is strange because we could receive the second Kinect's coordinate system in Rviz without more data. So I opened rqt_graph and found that it seemed like that both Kinects published the data on the same topic. So I doubt that if I need to apply distinct topic's name  to different cameras?  Or should I alter the launch file in another way?

2361098148, 2022-02-17:
**Thanks a lot! After your instruction, i have tried this way:**
<launch>
  <arg name="tf_prefix"         default="" />               

  <param name="robot_description"
    command="xacro $(find azure_kinect_ros_driver)/urdf/azure_kinect.urdf.xacro tf_prefix:=$(arg tf_prefix)" />

  <node name="joint_state_publisher1" pkg="joint_state_publisher" type="joint_state_publisher" />
  <node name="robot_state_publisher1" pkg="robot_state_publisher" type="state_publisher" />

  <arg name="depth_enabled"           default="true" />           
  <arg name="depth_mode"              default="NFOV_UNBINNED" />  
  <arg name="color_enabled"           default="true" />           
  <arg name="color_format"            default="bgra" />           
  <arg name="color_resolution"        default="3072P" />          
  <arg name="fps"                     default="15" />              
  <arg name="point_cloud"             default="true" />           
  <arg name="rgb_point_cloud"         default="false" />           
  <arg name="point_cloud_in_depth_frame" default="false" />        
  <arg name="required"                default="false" />          
  
  <arg name="recording_file"          default="" />               
  <arg name="recording_loop_enabled"  default="false" />          
  <arg name="body_tracking_enabled"           default="false" />  
  <arg name="body_tracking_smoothing_factor"  default="0.0" />    
  <arg name="rescale_ir_to_mono8"  default="false" />    
  <arg name="ir_mono8_scaling_factor"  default="1.0" />    
  <arg name="imu_rate_target" default="0"/>                        

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> 

    <param name="tf_prefix"         type="string" value="1" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000466592912" />       
    
    <param name="tf_prefix"         type="string" value="2" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

</launch>
I learned it from #97, but I met similar subsequent problems as well. It seemed like both Kinects worked, but only the first Kinect's data could be obtained. It is strange because we could receive the second Kinect's coordinate system in Rviz without more data. So I opened rqt_graph and found that it seemed like that both Kinects published the data on the same topic. So I doubt that if I need to apply distinct topic's name  to different cameras?  Or should I alter the launch file in another way?

2361098148, 2022-02-17:
<launch>
  <arg name="tf_prefix"         default="" />               

  <param name="robot_description"
    command="xacro $(find azure_kinect_ros_driver)/urdf/azure_kinect.urdf.xacro tf_prefix:=$(arg tf_prefix)" />

  <node name="joint_state_publisher1" pkg="joint_state_publisher" type="joint_state_publisher" />
  <node name="robot_state_publisher1" pkg="robot_state_publisher" type="state_publisher" />

  <arg name="depth_enabled"           default="true" />           
  <arg name="depth_mode"              default="NFOV_UNBINNED" />  
  <arg name="color_enabled"           default="true" />           
  <arg name="color_format"            default="bgra" />           
  <arg name="color_resolution"        default="3072P" />          
  <arg name="fps"                     default="15" />              
  <arg name="point_cloud"             default="true" />           
  <arg name="rgb_point_cloud"         default="false" />           
  <arg name="point_cloud_in_depth_frame" default="false" />        
  <arg name="required"                default="false" />          
  
  <arg name="recording_file"          default="" />               
  <arg name="recording_loop_enabled"  default="false" />          
  <arg name="body_tracking_enabled"           default="false" />  
  <arg name="body_tracking_smoothing_factor"  default="0.0" />    
  <arg name="rescale_ir_to_mono8"  default="false" />    
  <arg name="ir_mono8_scaling_factor"  default="1.0" />    
  <arg name="imu_rate_target" default="0"/>                        

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver1" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000319193812" /> 

    <param name="tf_prefix"         type="string" value="1" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

  <node pkg="azure_kinect_ros_driver" type="node" name="azure_kinect_ros_driver2" output="screen" required="$(arg required)">
    <param name="depth_enabled"     type="bool"   value="$(arg depth_enabled)" />
    <param name="depth_mode"        type="string" value="$(arg depth_mode)" />
    <param name="color_enabled"     type="bool"   value="$(arg color_enabled)" />
    <param name="color_format"      type="string" value="$(arg color_format)" />
    <param name="color_resolution"  type="string" value="$(arg color_resolution)" />
    <param name="fps"               type="int"    value="$(arg fps)" />
    <param name="point_cloud"       type="bool"   value="$(arg point_cloud)" />
    <param name="rgb_point_cloud"   type="bool"   value="$(arg rgb_point_cloud)" />
    <param name="point_cloud_in_depth_frame"   type="bool"   value="$(arg point_cloud_in_depth_frame)" />

    <param name="sensor_sn"         type="string" value="000466592912" />       
    
    <param name="tf_prefix"         type="string" value="2" />   

    <param name="recording_file"          type="string" value="$(arg recording_file)" />
    <param name="recording_loop_enabled"  type="bool"   value="$(arg recording_loop_enabled)" />
    <param name="body_tracking_enabled"           type="bool"   value="$(arg body_tracking_enabled)" />
    <param name="body_tracking_smoothing_factor"  type="double" value="$(arg body_tracking_smoothing_factor)" />
    <param name="rescale_ir_to_mono8" type="bool" value="$(arg rescale_ir_to_mono8)" />
    <param name="ir_mono8_scaling_factor" type="double" value="$(arg ir_mono8_scaling_factor)" />
    <param name="imu_rate_target" type="int" value="$(arg imu_rate_target)"/>
  </node>

</launch>

ooeygui, 2022-04-19:
Duplicating to the official support for multi-camera https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/126

closed 2022-04-19 by ooeygui