microsoft/Azure_Kinect_ROS_Driver #243: findk4abt.cmake cannot find body tracking SDK on Windows

sssphil, 2022-04-17:
**Describe the bug**
'findk4abt.cmake' is trying to find the dnn model with filename 'dnn_model.onnx' but there is no such file in body tracking SDK 1.1.0 and 1.1.1. I can see 'dnn_model_2_0_lite_op11.onnx' and 'dnn_model_2_0_op11.onnx' in the folder 'sdk/windows-desktop/amd64/release/bin'.

Default CMakeLists.txt file will only say 
```
!!! Body Tracking SDK not found: body tracking features will not be available !!!
```
But if I get rid of 'QUIET' on the find_package line, the terminal shows
```
CMake Warning at Azure_Kinect_ROS_Driver/cmake/FindModuleHelpers.cmake:7 (message):
  Rejecting SDK located at C:/Program Files/Azure Kinect Body Tracking SDK:
  Could not find dnn_model.onnx at C:/Program Files/Azure Kinect Body
  Tracking SDK/sdk/windows-desktop/amd64/release/bin/dnn_model.onnx
Call Stack (most recent call first):
  Azure_Kinect_ROS_Driver/cmake/Findk4abt.cmake:62 (quiet_message)
  Azure_Kinect_ROS_Driver/CMakeLists.txt:95 (find_package)

!!! Body Tracking SDK not found: body tracking features will not be available !!!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repo to catkin workspace
2. Run catkin_make in the workspace
3. See the first error
4. Change CMakeLists.txt from 'find_package(k4abt 1.0.0 QUITE MODULE)' to 'find_package(k4abt 1.0.0 MODULE)'
5. See the second error

**Expected behavior**
find the k4abt library (1.1.0 & 1.1.1) and build with body tracking features.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu]
 - Version [e.g. Version 1903, 18.04]

ooeygui, 2022-04-18:
The melodic build of the Azure Kinect is specifically built to use the [Body tracking SDK 1.0.0](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download). It appears that later builds of the body tracking SDK have changed names. To preserve compatibility with Melodic, I'm not going to change this for melodic. I'll create a noetic branch and move the codebase forward to the latest releases of the SDKs.

ooeygui, 2022-04-19:
This has been corrected in https://github.com/microsoft/Azure_Kinect_ROS_Driver/pull/244. Thanks for the report!

closed 2022-04-19 by ooeygui

sssphil, 2022-05-30:
Thanks for the update. But for anyone who might have problems with the version of body tracking SDK on NVIDIA RTX 3000 series cards, I found a work around to use newer body tracking SDK . First, change the model name in Findk4abt.cmake so it looks like `set(RELATIVE_WIN_DNN_MODEL_PATH "${RELATIVE_WIN_BIN_DIR}/dnn_model_2_0_op11.onnx")`, and then copy `directml.dll` from the SDK to everywhere `onnxruntime.dll` locates (not quite sure about this but it seem to work). The name was changed in v1.1.0 according to the change log [https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download](url).

I have a NVIDIA 3090 card and had some problems with older versions of SDK. Using the default BT SDK which is v1.0.1 blocks the node at initializing the body tracker. I think this has been mentioned in a few issues in azure kinect sdk's repo.