microsoft/Azure_Kinect_ROS_Driver #249: [k4a_ros_device_node]: Failed to poll cameras: node cannot continue.

AlezzzRd, 2022-06-02:
Hi,

My setups are ubuntu 20.04 and ros2 foxy. I used  `ros2 launch azure_kinect_ros_driver driver.launch.py` to launch the camera driver. In the beginning, the camera and the driver work fine. But like 1-2 minutes after the driver starts, The node will shut down automatically with the warning as follows. `[k4a_ros_device_node]: Failed to poll cameras: node cannot continue.` After the driver shuts down, I used `k4aviewer`, and I can see the camera is still running properly. Only the ros2 driver stops. Do you know what reasons could cause this problem? The driver shuts down every time I ran it.
[image]

Thanks a lot.

ooeygui, 2022-06-02:
Hi there,
The Kinect SDK isn't officially supported on Ubuntu 20.04. 
Can you share the instructions you used to install it?

AlezzzRd, 2022-06-02:
Hi,
I followed the instructions here.
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/foxy-devel/docs/building.md

ooeygui, 2022-06-02:
Thanks for the comment - We're working on official support on 20.04 for the SDK.

empty-spacebar, 2022-07-20:
> Thanks for the comment - We're working on official support on 20.04 for the SDK.
Hi,
Actually, it happens when i use ubuntu 18.04

empty-spacebar, 2022-07-20:
@ooeygui Hi, actually it happens when i use ubuntu 18.04

AlezzzRd, 2022-07-25:
Hi, finally I changed a laptop and this error did not happen again. Both of the laptops use ubuntu 20.04.

gyt290811, 2023-08-03:
@AlezzzRd @ooeygui @empty-spacebar @sverrejoh 
Hi, is this problem solved? I'm meeting the same problem!!

zhang159560293, 2024-04-14:
> @AlezzzRd @ooeygui @empty-spacebar @sverrejoh Hi, is this problem solved? I'm meeting the same problem!! ![2023-08-03 21-18-16屏幕截图](https://private-user-images.githubusercontent.com/92161272/258123455-1442df43-8d02-42c5-a2c7-1923d0610b43.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMwMDg5ODIsIm5iZiI6MTcxMzAwODY4MiwicGF0aCI6Ii85MjE2MTI3Mi8yNTgxMjM0NTUtMTQ0MmRmNDMtOGQwMi00MmM1LWEyYzctMTkyM2QwNjEwYjQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDEzVDExNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4YWVjYTNkYjQzZTA4YzBjYTk4MmIwZDJjZjkxNzgxNDc3YjZlYjJlNjlhNGM5MmJiMTI4YWEwMjVkYmQwODImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HoSrSNhsThnX0Lqwb4mT1f3yexFqZWhdiPEP_APaDK0)

I have solved this issue. Chang a usb3.2 gen2 line. It is because of usb port or usb line