microsoft/Azure_Kinect_ROS_Driver #256: Humble Support?

tonynajjar, 2022-10-28:
Does the foxy-devel branch support Humble? From what I tried, no. Here is (part of) the traceback I get when building:

```
root@logi-XMG-CORE-REN-M20:/code/ros2_ws# colcon build --packages-select azure_kinect_ros_driver
Starting >>> azure_kinect_ros_driver
[3.5s] [0/1 complete] [azure_kinect_ros_driver:build 80% - 3.1s]
[3.7s] [0/1 complete] [azure_kinect_ros_driver:build 80% - 3.3s]
[3.8s] [0/1 complete] [azure_kinect_ros_driver:build 80% - 3.4s]
--- stderr: azure_kinect_ros_driver                             
Finding K4A SDK binaries
!!! Body Tracking SDK not found: body tracking features will not be available !!!
K4A Libs: k4a::k4a;k4a::k4arecord
K4A DLLs: /usr/lib/x86_64-linux-gnu/libk4a.so.1.4.1;/usr/lib/x86_64-linux-gnu/libk4arecord.so.1.4.1
K4A Install Needed: FALSE
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp: In constructor ‘K4AROSDevice::K4AROSDevice()’:
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:61:26: error: no matching function for call to ‘K4AROSDevice::declare_parameter(const char [14])’
   61 |   this->declare_parameter("depth_enabled");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:61:26: note:   candidate expects 4 arguments, 1 provided
   61 |   this->declare_parameter("depth_enabled");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:434:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool)’
  434 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:434:3: note:   template argument deduction/substitution failed:
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:61:26: note:   couldn’t deduce template parameter ‘ParameterT’
   61 |   this->declare_parameter("depth_enabled");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:366:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  366 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:366:3: note:   candidate expects 4 arguments, 1 provided
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:391:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:62:26: error: no matching function for call to ‘K4AROSDevice::declare_parameter(const char [11])’
   62 |   this->declare_parameter("depth_mode");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:421:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool)’
  421 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:421:3: note:   template argument deduction/substitution failed:
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:62:26: note:   candidate expects 4 arguments, 1 provided
   62 |   this->declare_parameter("depth_mode");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:434:3: note: candidate: ‘template<class ParameterT> auto rclcpp::Node::declare_parameter(const string&, const ParameterDescriptor&, bool)’
  434 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:434:3: note:   template argument deduction/substitution failed:
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:62:26: note:   couldn’t deduce template parameter ‘ParameterT’
   62 |   this->declare_parameter("depth_mode");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:366:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, const rclcpp::ParameterValue&, const ParameterDescriptor&, bool)’
  366 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:366:3: note:   candidate expects 4 arguments, 1 provided
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:391:3: note: candidate: ‘const rclcpp::ParameterValue& rclcpp::Node::declare_parameter(const string&, rclcpp::ParameterType, const ParameterDescriptor&, bool)’
  391 |   declare_parameter(
      |   ^~~~~~~~~~~~~~~~~
/code/ros2_ws/src/rclcpp/rclcpp/include/rclcpp/node.hpp:391:3: note:   candidate expects 4 arguments, 1 provided
/code/ros2_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:63:26: error: no matching function for call to ‘K4AROSDevice::declare_parameter(const char [14])’
   63 |   this->declare_parameter("color_enabled");
      |   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
```

ooeygui, 2022-10-28:
Humble is not currently supported, but is on our backlog.

tonynajjar, 2022-10-28:
I got it to compile by adding defaults to the `declare_parameter()` functions. If you don't see much more effort to be put it to support humble, I can create a PR

ooeygui, 2022-10-28:
Thank you for the offer! I created a humble branch if you want to PR what you've done so far; that would definitely be helpful!
https://github.com/microsoft/Azure_Kinect_ROS_Driver/tree/humble

francescosemeraro, 2023-02-25:
Hi! Thanks for creating the branch. I am running ROS2 Humble on Ubuntu 22.04. When I try to build the package, I get the following error:

`Starting >>> azure_kinect_ros_driver
--- stderr: azure_kinect_ros_driver                             
/home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp: In member function ‘k4a_result_t K4AROSDevice::getBodyMarker(const k4abt_body_t&, std::shared_ptr<visualization_msgs::msg::Marker_<std::allocator<void> > >, int, rclcpp::Time)’:
/home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:771:47: error: no matching function for call to ‘rclcpp::Duration::Duration(double)’
  771 |   marker_msg->lifetime = rclcpp::Duration(0.25);
      |                                               ^
In file included from /opt/ros/humble/include/rclcpp/rclcpp/qos.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/node_interfaces/node_graph_interface.hpp:32,
                 from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:42,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:40,
                 from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:48:3: note: candidate: ‘template<class Rep, class Period> rclcpp::Duration::Duration(const std::chrono::duration<_Rep1, _Period1>&)’
   48 |   Duration(const std::chrono::duration<Rep, Period> & duration)  // NOLINT(runtime/explicit)
      |   ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:48:3: note:   template argument deduction/substitution failed:
/home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:771:47: note:   mismatched types ‘const std::chrono::duration<_Rep1, _Period1>’ and ‘double’
  771 |   marker_msg->lifetime = rclcpp::Duration(0.25);
      |                                               ^
In file included from /opt/ros/humble/include/rclcpp/rclcpp/qos.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/node_interfaces/node_graph_interface.hpp:32,
                 from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:42,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:24,
                 from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:40,
                 from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/include/azure_kinect_ros_driver/k4a_ros_device.h:15,
                 from /home/francesco/humble_ws/src/Azure_Kinect_ROS_Driver/src/k4a_ros_device.cpp:6:
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:151:3: note: candidate: ‘rclcpp::Duration::Duration()’
  151 |   Duration() = default;
      |   ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:151:3: note:   candidate expects 0 arguments, 1 provided
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:61:3: note: candidate: ‘rclcpp::Duration::Duration(const rclcpp::Duration&)’
   61 |   Duration(const Duration & rhs);
      |   ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:61:29: note:   no known conversion for argument 1 from ‘double’ to ‘const rclcpp::Duration&’
   61 |   Duration(const Duration & rhs);
      |            ~~~~~~~~~~~~~~~~~^~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:59:12: note: candidate: ‘rclcpp::Duration::Duration(const rcl_duration_t&)’
   59 |   explicit Duration(const rcl_duration_t & duration);
      |            ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:59:44: note:   no known conversion for argument 1 from ‘double’ to ‘const rcl_duration_t&’ {aka ‘const rcl_duration_s&’}
   59 |   explicit Duration(const rcl_duration_t & duration);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:53:3: note: candidate: ‘rclcpp::Duration::Duration(const Duration&)’
   53 |   Duration(const builtin_interfaces::msg::Duration & duration_msg);  // NOLINT(runtime/explicit)
      |   ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:53:54: note:   no known conversion for argument 1 from ‘double’ to ‘const Duration&’ {aka ‘const builtin_interfaces::msg::Duration_<std::allocator<void> >&’}
   53 |   Duration(const builtin_interfaces::msg::Duration & duration_msg);  // NOLINT(runtime/explicit)
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:42:12: note: candidate: ‘rclcpp::Duration::Duration(std::chrono::nanoseconds)’
   42 |   explicit Duration(std::chrono::nanoseconds nanoseconds);
      |            ^~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:42:46: note:   no known conversion for argument 1 from ‘double’ to ‘std::chrono::nanoseconds’ {aka ‘std::chrono::duration<long int, std::ratio<1, 1000000000> >’}
   42 |   explicit Duration(std::chrono::nanoseconds nanoseconds);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:39:3: note: candidate: ‘rclcpp::Duration::Duration(int32_t, uint32_t)’
[... cut ...]
/opt/ros/humble/include/rclcpp/rclcpp/duration.hpp:39:3: note:   candidate expects 2 arguments, 1 provided
gmake[2]: *** [CMakeFiles/azure_kinect_ros_driver_node.dir/build.make:90: CMakeFiles/azure_kinect_ros_driver_node.dir/src/k4a_ros_device.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/azure_kinect_ros_driver_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< azure_kinect_ros_driver [4.06s, exited with code 2]
`
Any idea of why this could be happening? I am not sure that is related to the fact I am using Humble though.

francescosemeraro, 2023-02-25:
Ok, I sorted it out. Apparently, you need to specify the duration in seconds and nanoseconds with two separate variables, while constructing the rclcpp::Duration object. Replacing this line with the highlighted one will do.

  `marker_msg->lifetime = rclcpp::Duration(int32_t(0),uint32_t(25000000));`

closed 2023-08-28 by tonynajjar