microsoft/Azure_Kinect_ROS_Driver #260: Having the driver node running breaks our URDF

dpt5216, 2022-11-30:
**Describe the bug**
We are running a system that uses ABB's ROS drivers to control an industrial arm for a photogrammetry system, with the kinect used for snapping photos

**To Reproduce**
You guys probably can't reproduce since you dont have our code. But when the driver is not running we get no errors. When it is running (on a different machine) our script which feeds trajectories to the ABB Driver fails setup.

**Expected behavior**
Successful setup

**Logs**
`
====initializing setup====
[ERROR] [1669764577.194687104]: Semantic description is not specified for the same robot as the URDF
[ERROR] [1669764577.196572448]: Virtual joint does not attach to a link on the robot (link 'base_link' is not known)
[ERROR] [1669764577.196614267]: Link 'base_link' declared as part of a chain in group 'manipulator' is not known to the URDF
[ WARN] [1669764577.196633957]: Group 'manipulator' is empty.
[ERROR] [1669764577.196667781]: Joint 'joint_1' declared as part of group state 'all-zeros' is not known to the URDF
[ERROR] [1669764577.196684872]: Joint 'joint_2' declared as part of group state 'all-zeros' is not known to the URDF
[ERROR] [1669764577.196699856]: Joint 'joint_3' declared as part of group state 'all-zeros' is not known to the URDF
[ERROR] [1669764577.196713929]: Joint 'joint_4' declared as part of group state 'all-zeros' is not known to the URDF
[ERROR] [1669764577.196728393]: Joint 'joint_5' declared as part of group state 'all-zeros' is not known to the URDF
[ERROR] [1669764577.196742533]: Joint 'joint_6' declared as part of group state 'all-zeros' is not known to the URDF
[ WARN] [1669764577.196769339]: Link 'base_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196785027]: Link 'base_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196799355]: Link 'base_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196813608]: Link 'link_1' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196827682]: Link 'link_1' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196848268]: Link 'link_2' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196862570]: Link 'link_2' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196876401]: Link 'link_2' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196890233]: Link 'link_2' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196904194]: Link 'link_3' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196917876]: Link 'link_3' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196931666]: Link 'link_3' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196945260]: Link 'link_4' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196958548]: Link 'link_4' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1669764577.196972122]: Link 'link_5' is not known to URDF. Cannot disable/enable collisons.
[ INFO] [1669764577.197014461]: Loading robot model 'azure-kinect'...
[ INFO] [1669764577.197034611]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1669764577.197054155]: Link camera_base has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1669764577.197074962]: Link camera_body has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1669764577.197091050]: Link camera_visor has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1669764577.197121011]: No geometry is associated to any robot links
[ WARN] [1669764577.197135904]: Group 'manipulator' must have at least one valid joint
[ WARN] [1669764577.197148301]: Failed to add group 'manipulator'
[ERROR] [1669764577.197162135]: Group state 'all-zeros' specified for group 'manipulator', but that group does not exist
[FATAL] [1669764577.251637278]: Group 'manipulator' was not found.
`

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

**Additional context**
We have no reason to load the azure-kinect robot model for our purposes, and are very surprised that it tries to load automatically. This, and what it does to our groups, seems to be causing the issue. Is there any way to simply turn this off?

ooeygui, 2022-11-30:
I'm not sure what the ask is - can you clarify?
I suspect there's a name collision. Can you produce a URDF file which exhibits the behavior even without the ABB arm?

dpt5216, 2022-11-30:
We managed to fix it by running the kinect last

closed 2022-11-30 by dpt5216