microsoft/Azure_Kinect_ROS_Driver #271: how to set trigger mode of Azure kinect DK in this rospackage via MCU

gyt290811, 2023-08-03:
Hi, 
we were trouble in setting `wired_sync_mode` in `kinect_rgb.launch` to make sure Azure Kinect camera synchronization.

Do you know any method to solve this problem
[image]

ooeygui, 2023-08-03:
Hello @gyt290811,
Multiple camera support in the ROS node has been a long-standing request but has not been funded. 

However, I do see that in the screenshot, the computer you are using appears to be a NUC. The computer is likely overwhelmed. The Kinect camera very specific requirements on hardware - Each Kinect should be on its own USB bus in order to prevent contention.

gyt290811, 2023-08-04:
Hi, @ooeygui 
I am using an MCU external hardware to trigger the Azure Kinect DK（only one camera) in the `Sub external sync`(slaver mode), and it can be successfully triggered in the SDK. When using the [official ROS package](https://github.com/microsoft/Azure_Kinect_ROS_Driver) without modifying anything, it can run. However, after changing the `wired_sync_mode=2` in `kinect_rgb.launch`, the ROS package cannot run properly and and I encounter the error shown in the screenshot.