microsoft/Azure_Kinect_ROS_Driver #272: [FATAL] [1691069372.183638523]: Failed to poll cameras: node cannot continue.

gyt290811, 2023-08-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
Please provide relevant logs to help diagnose your issue. To assist in debugging, please enable *info* level logging of the Azure Kinect Sensor SDK and attach any logs you have to this issue.
See [k4atypes.h](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/feecae0456511ac734287571b101d10fd7292673/include/k4a/k4atypes.h#L184) for how to enable logs.

Please also enable *DEBUG* level logging of the ROS node.
See [roscpp/Overview/Logging](http://wiki.ros.org/roscpp/Overview/Logging) for how to enable ROS DEBUG logs.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [ Ubuntu]
 - Version [20.04]

**Additional context**
Add any other context about the problem here.
[ WARN] [1691069368.183421662]: Camera calibration file /home/yc/.ros/camera_info/azure_kinect_rgb_000863414212_1536P.yaml not found.
[ INFO] [1691069368.183456550]: using default calibration URL
[ INFO] [1691069368.183473747]: camera calibration URL: file:///home/yc/.ros/camera_info/azure_kinect_ir_000863414212_NFOV_UNBINNED.yaml
[ INFO] [1691069368.183495708]: Unable to open camera calibration file [/home/yc/.ros/camera_info/azure_kinect_ir_000863414212_NFOV_UNBINNED.yaml]
[ WARN] [1691069368.183515008]: Camera calibration file /home/yc/.ros/camera_info/azure_kinect_ir_000863414212_NFOV_UNBINNED.yaml not found.
[ INFO] [1691069368.183793842]: IMU started
[FATAL] [1691069372.183638523]: Failed to poll cameras: node cannot continue.
[k4a_nodelet_manager-4] process has finished cleanly
log file: /home/yc/.ros/log/c3e91d08-3201-11ee-922d-292bbb9d97a8/k4a_nodelet_manager-4*.log

zhang159560293, 2024-04-13:
I meet this issue, too. In future if it will be solved , could some one can reply @me. Very Very Very thanks.

zhang159560293, 2024-04-14:
> I meet this issue, too. In future if it will be solved , could some one can reply @me. Very Very Very thanks.

I have solved this issue. Chang a usb3.2 gen2 line. It is because of usb port or usb line

ooeygui, 2024-04-17:
This ROS node is built on the Azure Kinect SDK. The SDK is not supported on Linux beyond 18.04 (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1263). With the Azure Kinect DK being end of life, the sdk will not be supported on 20.04 or later.