microsoft/Azure_Kinect_ROS_Driver #274: Installation problem

simahsa, 2023-09-12:
Hi, I am trying to get the Azure_Kinect_ROS_Driver working on Windows 10. I have installed ROS and Azure Kinect Sensor SDK successfully and followed the steps catkin make and ... but when I enter "source devel/setup.bash I am getting the following error
'source' is not recognized as an internal or external command,
operable program or batch file.

I am new in this field, can anyone help me please?

closed 2023-09-12 by ooeygui

simahsa, 2023-09-12:
> <microsoft/Azure_Kinect_ROS_Driver/issue/274/issue_event/10350282070@
> github.com>

I am still getting the same error:
'devel' is not recognized as an internal or external command, operable
program or batch file.

ooeygui, 2023-09-12:
Sorry, I had a typo. `devel\setup.bat`

simahsa, 2023-09-13:
Thank you! It worked. (I did use call devel\setup.bat (source command in
Linux))
Then I proceed with this command: roslaunch azure_kinect_ros_driver but I
get this error: [31mRLException: [azure_kinect_ros_driver] is not a launch
file name [0m
[31mThe traceback for the exception was written to the log file [0m
Can you help me to fix it please?